plasmapper/modbus-esp-cpp #3: "PRO CPU has been reset by WDT" when reading more registers at a time

MichaelUray, 2022-11-20:
I am experiencing a watchdog issue, when I am reading more registers (e.g. 80) at once.

On my PC is the [Modbus Simulator](https://sourceforge.net/projects/modrssim/) running and it is connected via an USB serial/TTL converter to my ESP32.
To read 40 registers at once (`uint16_t holdingRegisters[40]`) works fine, but as soon as I change it to 80 registers (`uint16_t holdingRegisters[80]`), the watchdog triggers and the ESP reboots.

This is my testing code:
```C
#include "pl_modbus.h"

extern "C" void app_main(void)
{
  auto port = std::make_shared<PL::UartPort>(UART_NUM_1, 256, 256, 10, 9, -1, -1);
  PL::ModbusClient client(port, PL::ModbusProtocol::rtu, 1);

  port->Initialize();
  port->SetBaudRate(115200);
  port->SetDataBits(8);
  port->SetParity(PL::UartParity::none);
  port->SetStopBits(PL::UartStopBits::one);
  port->SetFlowControl(PL::UartFlowControl::none);
  port->Enable();

  while (1)
  {
    uint16_t holdingRegisters[80];
    if (client.ReadHoldingRegisters(0, sizeof(holdingRegisters) / sizeof(uint16_t), holdingRegisters, NULL) == ESP_OK)
    {
      for (int i = 0; i < sizeof(holdingRegisters) / sizeof(uint16_t); i++)
        holdingRegisters[i]++;
    }
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
}
```

The serial log:
```
I (105) boot: Partition Table:
I (108) boot: ## Label            Usage          Type ST Offset   Length
I (116) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (123) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (131) boot:  2 factory          factory app      00 00 00010000 00100000
I (138) boot: End of partition table
I (143) boot_comm: chip revision: 3, min. application chip revision: 0
I (150) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0e264h ( 57956) map
I (179) esp_image: segment 1: paddr=0001e28c vaddr=3ffb0000 size=01d8ch (  7564) load
I (183) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=2c6c8h (181960) map
I (251) esp_image: segment 3: paddr=0004c6f0 vaddr=3ffb1d8c size=003dch (   988) load
I (252) esp_image: segment 4: paddr=0004cad4 vaddr=40080000 size=0c338h ( 49976) load
I (277) esp_image: segment 5: paddr=00058e14 vaddr=50000000 size=00010h (    16) load
I (284) boot: Loaded app from partition at offset 0x10000
I (284) boot: Disabling RNG early entropy source...
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6660
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (58) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (58) boot: compile time 18:34:05
I (58) boot: chip revision: 3
I (61) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (68) boot.esp32: SPI Speed      : 40MHz
I (73) boot.esp32: SPI Mode       : DIO
I (77) boot.esp32: SPI Flash Size : 4MB
W (82) boot.esp32: PRO CPU has been reset by WDT.
W (87) boot.esp32: WDT reset info: PRO CPU PC=0x400862e8
0x400862e8: _xt_context_save at C:/Users/Michael.Uray/esp/esp-idf-v4.4.2/components/freertos/port/xtensa/xtensa_context.S:209

W (93) boot.esp32: WDT reset info: APP CPU PC=0x400d1f3e
0x400d1f3e: panic_handler at C:/Users/Michael.Uray/esp/esp-idf-v4.4.2/components/esp_system/port/panic_handler.c:141 (discriminator 1)

I (99) boot: Enabling RNG early entropy source...
I (105) boot: Partition Table:
I (108) boot: ## Label            Usage          Type ST Offset   Length
I (116) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (123) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (131) boot:  2 factory          factory app      00 00 00010000 00100000
I (138) boot: End of partition table
I (143) boot_comm: chip revision: 3, min. application chip revision: 0
I (150) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0e264h ( 57956) map
I (179) esp_image: segment 1: paddr=0001e28c vaddr=3ffb0000 size=01d8ch (  7564) load
I (183) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=2c6c8h (181960) map
I (251) esp_image: segment 3: paddr=0004c6f0 vaddr=3ffb1d8c size=003dch (   988) load
I (252) esp_image: segment 4: paddr=0004cad4 vaddr=40080000 size=0c338h ( 49976) load
I (277) esp_image: segment 5: paddr=00058e14 vaddr=50000000 size=00010h (    16) load
I (284) boot: Loaded app from partition at offset 0x10000
I (284) boot: Disabling RNG early entropy source...
```

At this point the watchdog happens as soon as I start the Modbus slave simulator on my PC and it returns some data to the ESP:
```
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
```

The `defaultBufferSize` in the library is set 260 bytes, which should be enough to read 80 registers (160 bytes + a bit overhead).

Any idea what could cause this issue?
Maybe any other buffers which are defined too small?

plasmapper, 2022-11-21:
I could not reproduce your problem. I compiled your code, tested it with [Modbus Simulator](https://sourceforge.net/projects/modrssim/) and it worked fine.
And also I can't see anything about Modbus code in the error log. It seems like our system restarts when it is supposed to start the CPUs, long before the Modbus code.

MichaelUray, 2022-11-24:
It looks it was somehow a hardware related issue, when I connect the JTAG debugger the bevaiour changes.
Not sure what it was exectly, but I guess it has something to do that I changed the USB/serial adapter to another type as I used before without problems. Maybe because the ESP32 gets supplied by an USB hub with an external supply and the serial converter directly from my mainboard.

Sorry for bothering you here with issues which actually have nothing to do with your library.

closed 2022-11-24 by MichaelUray