plasmapper/modbus-esp-cpp #4: How to handle RS485 DIR

NemesisXB, 2024-07-12:
How do you handle the RS485 half duplex DIR pin?

plasmapper, 2024-07-12:
To use Modbus client over the RS-485 line you should use [this constructor](https://plasmapper.github.io/esp-cpp/components/modbus/docs/api/modbus_client.html#_CPPv4N2PL12ModbusClientE). The first parameter of this constructor is the pointer to the UART object which is created using [this constructor](https://plasmapper.github.io/esp-cpp/components/uart/docs/api/uart.html#_CPPv4N2PL4UartE). It has the rtsPin parameter that you can use to specify the pin that is connected to the ~RE and DE pins of your RS-485 driver chip.
You can try it with [this example](https://components.espressif.com/components/plasmapper/pl_modbus/versions/1.0.1/examples/uart_client?language=en).
Just replace
<pre>
auto uart = std::make_shared<PL::Uart>(UART_NUM_0);
</pre>
with
<pre>
auto uart = std::make_shared<PL::Uart>(UART_NUM_0, PL::Uart::minBufferSize, PL::Uart::minBufferSize, <b>[TX pin]</b>, <b>[RX pin]</b>, <b>[RTS pin]</b>);
</pre>

NemesisXB, 2024-07-13:
Thanks for the quick response. I will be trying this out first thing Monday.

NemesisXB, 2024-07-17:
Thanks. This worked. I did need to add this to the example for it to work:

closed 2024-07-25 by NemesisXB